starryalley/Anduril2 #1: UI Flow Chart

AutomateMyLifee, 2021-08-02:
I updated the Anduril 2 flow chart to reflect the changes you made as I had a difficult time visualizing them. Hope this helps others.
[image]

starryalley, 2021-08-02:
This is so awesome! I sometimes need to check what I've done through my README but this is great. Thanks so much for your work!

AutomateMyLifee, 2021-08-02:
You're welcome. Thanks for your code. It's fun to get as many features as I can. I got it complied for my D4SV2 with W2 emitters and it works great! Only I couldn't get it to work with my D4V2 with 219C emitters. It has the kr4 driver but the 219c so FET power has to be limited to 80% power. I compiled using the config noctigon-kr4-219, but the trouble is that as it ramps up, once it reaches a high brightness (not sure if it's the max), the light just stops shining. When I ramp down a bit it turns back on. Any idea what this could be?

starryalley, 2021-08-02:
This sounds very familiar. I had this happened once on my D4v2 KR4 driver light as well but I couldn't remember now how I solved this. This is definitely code issue that I did not foresee. My guess is that this is related to the ramp setting. I've changed the ramp length to 200 for KR4 and KR4-nofet light so maybe you also need to change that in kr4-219?

I will take a look this week and update here. Thanks for the feedback!

I have D4v2 219b lights coming so I might also need to update the code base to ToyKeeper's latest revision to include the 219b config.

starryalley, 2021-08-03:
@AutomateMyLifee your issue on 219 light is definitely about ramp length and the ramp levels. In cfg-noctigon-kr4-219.h it uses cfg-noctigon-kr4.h, where I've changed `RAMP_LENGTH` to `200`. But in kr4-219 `PWM1_LEVELS` and `PWM2_LEVELS`weren't changed (which has 150 levels). 

You can either use the `level_calc.py` tool to generate 200-length levels, or define `RAMP_LENGTH` back to 150 (Also please change related defines such as `MAX_1x7135`, `RAMP_SMOOTH_CEIL` and `THERM_FASTER_LEVEL` etc.

You can also wait a few days (or weeks?) after I make those changes for my coming 219b lights.

starryalley, 2022-10-11:
@AutomateMyLifee thanks for the flow chart again. I want to make some changes to the flow chart since I've changed things a bit in the repo now. Is it possible that you share the flow chart's original file so I can also make some changes? Thank you.

AutomateMyLifee, 2022-10-13:
The file I already shared is the highest quality I have.

starryalley, 2022-10-13:
Thanks very much @AutomateMyLifee I'll see what I can do :)