starryalley/Anduril2 #3: Exceeds memory limit

flashquark, 2022-09-20:
Whenever I try to compile your fork, it says that "GccApplication4.elf section `.text' will not fit in region `text'" and that it overflowed by 1534 bytes. By any chance do you remove anything before you compile your hex files, or is there something that might be causing something extra to compile on my side? ToyKeeper's original code compiles just fine for me.

starryalley, 2022-09-20:
Hi @flashquark I added many things to the codebase and I only have D4v2, SP10 pro and TS10 where there are still plenty of spaces available. Not sure what target you are building but yes that error can happen for other lights. But if you are building for D4v2 (single or dual channel), sp10 pro, ts10, then that error should not happen. 

I use Mac and Ubuntu to build the hex. 

To reduce space, the quickest way is to disable some unused strobe modes in `config-default.h`

```
#define USE_BIKE_FLASHER_MODE
#define USE_PARTY_STROBE_MODE
#define USE_TACTICAL_STROBE_MODE
#define USE_LIGHTNING_MODE
#define USE_CANDLE_MODE
#define USE_FIREWORK_MODE
```

or in the each light's config.

flashquark, 2022-09-21:
I’m building for a dual channel DT8 which interestingly from 15 clicks, shows that it uses the same hex as the D4SV2 tint ramp without fet. (That’s interesting as I would have assumed that it would have used the fet version since my DT8 doesn’t have E21 but rather, has Osram W2 and LH351D emitters that should be able to withstand fet.) 

In any case, I did manage to get it compiled by disabling unused strobes like you suggested. I am building on Windows, so perhaps there are some slight differences with the libraries and what gets included during compile time. Other than disabling the unused strobes, I also had to change the pseudorand function that you used for start up modes into rand instead.

Thanks for the help!

closed 2022-09-21 by flashquark

starryalley, 2022-09-22:
It's weird to have different build under Windows. I have no idea about the toolchain in Windows though. Glad that you resolved it.