starryalley/Anduril2 #5: Ramping Bug/Issue?

BrightZin, 2022-11-03:
I flashed the latest kr4 no fet hex on 2 e17a d4v2's and I noticed there's a fluttery type of double blink happening when smooth ramping  up through mid ramp. Not sure it it was intentional or not but I don't think non fet anduril is supposed to blink mid ramp.

starryalley, 2022-11-03:
Yes I noticed this as well on my D4v2 lights but I haven't got time to figure out. Might be a bug or a configuration issue. I'll let you know when I find out why. Thanks!

starryalley, 2022-11-03:
Doing some quick bisecting and found the faulty commit: [`1923dd27ae3be10f48dc7926dc1b647815a78071`
](https://github.com/starryalley/Anduril2/commit/1923dd27ae3be10f48dc7926dc1b647815a78071)

Apparently I forgot to enable the phase sync config for KR4 (all variants). Fixed with [this commit](https://github.com/starryalley/Anduril2/commit/48b9b3991152a8475a08f2cf38fdaff9a3ac6ea0)

Thanks again for brining this to my attention. I'll create a new release shortly.

BrightZin, 2022-11-03:
That was quick. Thanks for your work on this. Gonna flash some more lights once when its released. That blink was really annoying me

starryalley, 2022-11-03:
I agree. I haven't EDC'd my D4v2 for the past few months (I've been using my SP10 pro and TS10 lately). All my D4v2's are stationary at home so that's my excuse not to have a motivation to analyse behind that annoying blink. Thanks again for this issue or I might never get it fixed.

I've personally verified on 4 of my D4v2 currently on my desk.

Just created a new release [here](https://github.com/starryalley/Anduril2/releases/tag/release_20221103)

BrightZin, 2022-11-04:
I flashed the no fet kr4 version and the flickering during ramping is gone but now the light flickers right when I turn it on at mid ramp or higher

starryalley, 2022-11-04:
Any chance it is the startup mode default being set to ramp up (mode 1)? 
See details: https://github.com/starryalley/Anduril2#start-up-mode

Set to mode 0 for the default Anduril2 behaviour. I prefer mode 1 so that's why I set it in [this commit](https://github.com/starryalley/Anduril2/commit/6fcb0f7c79f0491abc8784b7ffe2ff02ace7d303)

BrightZin, 2022-11-04:
No I just checked and I'm in mode 0. It turns on instantly but flickers the same way it used to during ramping

starryalley, 2022-11-04:
Sure thanks for reporting. I'll see what's wrong there! SP10 pro doesn't seem to have it so I assume it's another missing part/config. I'll update when I find out.

starryalley, 2022-11-04:
I can replicate the issue on my nofet and fet D4v2. This is clearly the phase sync issue, which results in the same flickers of the original problem like in the ramping up. But this is not the same as the previous fix. This issue actually exists before the latest release.

It is a bug in my startup mode code introduced in [this commit](https://github.com/starryalley/Anduril2/commit/c4149b8c0abdc5804840baa8f9193c0bd7057a95). 

Fix [here](https://github.com/starryalley/Anduril2/commit/71fd40f1bffa85af18269ab5bd73fc86656a8ce4).

I've attached kr4-nofet HEX. If you can give it a try and let me know. Thanks again for the report and test!

[anduril.noctigon-kr4-nofet.hex.zip](https://github.com/starryalley/Anduril2/files/9937882/anduril.noctigon-kr4-nofet.hex.zip)

BrightZin, 2022-11-04:
Looks good, no more blinking on start up. Thanks

starryalley, 2022-11-05:
Thanks again for the bug report and testing. I've just published another [release](https://github.com/starryalley/Anduril2/releases/tag/release_20221105).

Closing.

closed 2022-11-05 by starryalley