starryalley/Anduril2 #6: Bug: D4SV2 - Turbo and Strobe Modes crashing

Adrian-Grimm, 2022-11-10:
D4SV2:
From Ramp 2C to Turbo crashed with a last bright blink.
By entering Strobe Modes it's quite the same bright blinkt and then off.
PowerReset helps always.

starryalley, 2022-11-10:
Thank you for the report and test. 

Actually I don't have a D4sV2 so this mod isn't tested or even considered for D4sV2. Did you build the hex by yourself? The emisar-d4sv2-tintramp* hex is for my D4V2 tint ramping lights since they came preinstalled with this variant.

I'll try to see if there is anything missing there.

Adrian-Grimm, 2022-11-10:
My D4SV2 came with "0135 emisar-d4sv2-tintramp" so I thought your hex file fits ... I mean everything is working perfect except of this bug..

Maybe I find some time to dig deeper into all of this.

starryalley, 2022-11-17:
I just took a look at the cfg and I couldn't see why this happens. Are you using [anduril.emisar-d4sv2-tintramp-fet.hex](https://github.com/starryalley/Anduril2/releases/download/release_20221105/anduril.emisar-d4sv2-tintramp-fet.hex)? If so, can you try using the non-fet version? I actually don't have anything that runs tintramp-fet so I probably shouldn't have included it in the release.

Does 2C from ramp goes to level 150 or 130 in your light? Maybe 2C from off and 2C from on go to different levels in your light and one of the level's config is wrong. I also have no idea if fet enabled tint ramping lights' drivers are the same as my  non-fet tint ramping ones. My guess is maybe they are also the same, just that mine has fet disabled in the firmware because mine has E21a and 219b LEDs. But maybe I'm wrong. 

Sorry can't help here since I don't have a D4SV2. Please let me know if you also find something. Thank you.

Adrian-Grimm, 2022-11-18:
I got it more figured out what's the issue. 
If I my battery is fully charged or above 90% - I have this error (XTAR 26650 5200mAh).
But as soon as the level reaches 90% or lower everything is working perfect.
Strange...

starryalley, 2022-11-21:
That's a good finding. Can you try with stock Anduril2 firmware (latest) and the same battery fully charged? I've compared the config it is basically the same as stock near the top of the ramp (I've modified a bit in the lowest few levels in the ramp).

How about using the non-fet firmware, is everything good then?

Adrian-Grimm, 2022-12-05:
@starryalley With both versions the same Bug, with a full charged battery (Non-fet, -fet).
I compared the "anduril.2021-12-13.emisar-d4sv2-tintramp-fet.hex" to them which don't have that bug, but:

> There is a difference in the behavior with 2C from OFF to Turbo to your Versions: it first turns on with a lower brightness for a half second and then it goes to the full power.

I could imagine that this level step is required for some reason.

starryalley, 2023-02-23:
Sorry but I revisited this issue today and I couldn't find anything in the code that can turns on with a lower brightness for 0.5s and then goes to full power (FET).  Maybe I didn't look very carefully. 

Without the actual hardware it's really hard to debug. I excluded the -fet build in today's release so maybe one day when I feel like getting a D4sv2, I will check this.

Adrian-Grimm, 2023-02-23:
@starryalley I think you should not exclude this build - it's not that big of an problem. And I really like your version even with this little bug.

starryalley, 2023-02-24:
Added that hex back into the latest release :)

Adrian-Grimm, 2023-02-27:
OMG - somehow the issue is solved with the new version - I don't know if you really changed nothing - but it's gone... 👍

starryalley, 2023-02-27:
Haha good to hear. I didn't change anything specific to this build. Let's close this unless you see this again. Thanks for the bug report!

closed 2023-02-27 by starryalley