starryalley/Anduril2 #10: IDEA: Any kind of voltage indicator when main emitter on

webshadow, 2023-02-19:
Is it possible with RGB Button or blinking? I'm assuming there is a hardware limitation but since you already allowed switch to light up while light is active ...

starryalley, 2023-02-19:
There is a separate control to the button LED so yes we can use blinking to do some kind of voltage indication. I will have my first Hank light with RGB button as well so I can probably play around that. I have no idea how RGB button works so I will have to test this as well.

However what kind of voltage indication do you think is reasonable/practical when main emitter(s) are on?
- keep on showing current voltage via blinking (single color button LED), short pause like 3s and keep blinking?
- only blinks when low voltage?
- show different color based on current voltage (color button LED)?

webshadow, 2023-02-19:
I think it would be best to have a choice, RGB with the colors we already
know or blinking the voltage every time you loose 0.1v perhaps even have
that configurable? I feel like that's a nice tradeoff between info and calm
operation so to speak. Based on the time it takes for the voltage to
decrease we could even dream about an estimated time left? Sorry for going
of the rails there 😅

starryalley, 2023-02-20:
Estimated time left? That's something cool! Imagine I set the light to my desired output and use it as ceiling bounce. Then some 5 min later, I click maybe `11C` and I get a 9 blinks, meaning there is ~90min left before the voltage drops to 3.2V or something.

But that estimate might be way off based on voltage drop (is it linear drop?) and thermal step down or something. But it maybe is a good guess than nothing.

About the voltage indicator let me play around with my new light with RGB button and come back.

starryalley, 2023-02-20:
Unfortunately with RGB button on a D4K or D4v2:
```The Aux and the RGB switch are wired together on lights with front Aux, you could get your D4K to behave like your D1v2 but the front aux will stay on with the main emitters like the switch```

quoted from [here](https://www.reddit.com/r/flashlight/comments/z5hsnc/comment/ixw5ckj/?utm_source=share&utm_medium=web2x&context=3)

That means there is no way to separately control the button LED. If we are going to use button LED to indicate something, the front facing AUX LED will light up as well!

webshadow, 2023-02-21:
So that's all, i'm greatly relieved, i think this is absolutely acceptable
for what you get. I read something about hardwired too but thought it means
only the driver on the D1 one could physically do certain things with the
button. The way i see it with this function and the rest that you've
already done we almost have anduril 3 on our hands. I would show this off
on all of my lights and i'm sure adoption of your fork will get a big boost
!

starryalley, 2023-02-23:
Thanks for the kind words. I really appreciate it. 

I am thinking maybe I will create a new config specific for color button LED (for example, anduril.emisar-d4sv2-tintramp-color-button.hex) so when ramping up to a specific level (maybe level 20 or 40?) where turning on front facing AUX won't affect a thing, I can then turn on the AUX (along with the color button LED) and do the voltage indicator or temperature indicator. I miss the lighted button when emitters are on!

webshadow, 2023-02-25:
Sounds great, just tested AUX + about level 20 main and could easily spot
AUX Color (worst case > red) , got much worse at 40.