starryalley/Anduril2 #9: IDEA: Hybrid Tintramp/Instant Channel Switching

webshadow, 2023-02-19:
What about a third option in the ramping config to have both. 3C Switch/3H Ramp for example.

starryalley, 2023-02-19:
That's a good idea. Last week I just integrated SammysHP's [stepped tint ramping](https://github.com/SammysHP/flashlight-firmware/tree/stepped-tint-ramping) to this fork and found it very useful. 

In a normal smooth ramping (option 0), it currently has shortcut to go to middle tint (8H). Maybe I'll add another shortcut to go to both extremes. 8H doesn't sound like a "short"cut but I can't find anything with less clicks. I'll see.

starryalley, 2023-02-20:
I've thought about this:

- in normal mode (emitters on), `5C` to go to tint edge, and then `5C` to go to the other edge. `5C` is already used in candle, lightning and fireworks mode so this instant channel switching won't work in those strobe modes.
- `8H` to go to middle tint (already there). This works in all modes including strobe modes.

I can't use `3C` because `3C` is already defined in strobe mode to go to the previous mode, and in normal (steady) mode `3C` is used to increase brightness by 1 step. 

This goes along with smooth tint ramping or the stepped tint ramping. No matter what tint ramping style you are in. It will also work if you set the tint style to 1-mode (middle tint) only. (`3H` again and it will revert to middle tint in that case)

I can't decide if I should swap the functionality of `3C` and `5C` based on what's used more often. I've got 3 tint ramping lights now so it is getting used a bit more and more for myself.

I hope this works. Let me know what's your thoughts?

webshadow, 2023-02-21:
From the top of my head i'd say 5C sounds good, having ICS in the strobe
modes seems a little overkill.

webshadow, 2023-02-21:
Perhaps you could get rid of 3C in normal mode if the ramping had a
logarithmic function (configurable in ramp conf), that is fine at the
beginning and getting more coarse to the end.

starryalley, 2023-02-23:
`5C` shortcut to go to tint edges is included in the [latest release](https://github.com/starryalley/Anduril2/releases/tag/release_20230223)

I'll think about that log ramp thing. Sounds like a good one but maybe not practical at mid to higher brightness range?

starryalley, 2023-02-23:
I'll close this one and track the log ramp thing in #11 . Thanks for this great idea. I'm using `5C` a lot on my 3 dual channel lights.

closed 2023-02-23 by starryalley