starryalley/Anduril2 #20: Channel switching strobe modes have issues at ramp levels >130

SiteRelEnby, 2023-03-24:
By default, I can ramp brightness up to 150/150 on the dual channel strobe modes (even with normal ceiling set to 130), at which point the two channels start being mixed all the time as a steady brightness (effectively just 200% turbo).

The effect is actually quite nice for the smooth alternating mode at levels above 130 (it still alternates, but neither channel ever switches fully off) but at level 150 it is just 200% turbo with no channel switching. The best idea might be to pause ramping up at level 130 and blink once, then if the user keeps holding the button after a second or two, continue up to 149 and stop there. The alternative of stopping at the configured ceiling is the other logical way, but I like that one less as I have my normal ceiling at 130/150 on dual channel lights (since my anduril fork has direct shortcuts to 200% and both 100% turbo modes all available at the same time) and still like the effect of the channel ramping modes at 130-149, 150 is just obviously unintentional-appearing. Alternatively, making which it does be user-configurable.

On the other hand, the instant alternating strobe mode works less well with this - if channels are mixed in ramp mode and then that strobe is selected, it will alternate between the *mixed* channels and the second channel on 100%. The two approaches I can think of to handle this would be to either always set to single channel max at each end, or to invert the ramp (e.g. if the user ramped to a tint value of 64, alternate between that 191 as the opposite point (easy to obtain with `tint ^ 0xFF`).

May hack on this myself a bit if I have time. Really like your work, have incorporated some of your changes into my fork as well.

starryalley, 2023-03-24:
Thank you for this feedback and proposed solution! I'll see what I can do when I have some time. 
Thanks for using this fork too. Feel free to take whatever you like to yours as well!

SiteRelEnby, 2023-03-30:
Fixed it in my version, backported: https://github.com/starryalley/Anduril2/pull/21

starryalley, 2023-04-01:
Awesome, thank you! I will need to find some time to check this!

starryalley, 2023-04-18:
Hi @SiteRelEnby I finally got some time to review this. My apologies. I have got 2 busy little girls to deal with every day after work so less time now for Anduril2. Thanks for the excellent fix in this PR. Your code works perfectly. 

However on my (limited numbers of) dual channel lights I don't see a benefit of going above RAMP_SMOOTH_CEIL (130) even in smooth alternating mode. Maybe it works better in some LED combinations. I hope you don't mind I won't merge the PR. I will however adopt your proposed fix so the max level user can go is 130, for both alternating modes.

Your fork that has direct shortcuts to 200% and both 100% turbo modes is brilliant. Since I'm getting more dual/multi channel Anduril2 lights,  I might copy that over here :P 

Thank you again.

closed 2023-04-18 by starryalley

SiteRelEnby, 2023-04-19:
Yeah, no problem there, I like it with my white/red light although it's less useful on flood/throw. I'm keeping my fix in my version, or maybe make it configurable once I'm finished with LT1S Pro support.

commit 1cce9f43 (2023-04-18): strobe-modes: limit tint alternating mode max to ramp ceil

It used to be MAX_LEVEL (150) so when it goes above ramp ceiling (130)
it will actually activates both channels. Limit the max level both
tint alternating modes can go to the ramp ceiling.

Fix #20