starryalley/Anduril2 #22: K9.3/DM1.12 support

OKflashlightaholic, 2023-03-31:
Please add a k9.3/ dm1.12 hex support. The dual channel d4sv2 hex works but it enables 200% turbo which doesn't play well with Nichia e21a 3500k and deep red sst20's. The regular hex only turbos the fet channel. Thank you so much for all your hard work on your custom anduril 2 branch. I greatly appreciate the dual channel fireworks mode and the adjustable candle/fireplace mode and lighting mode.

starryalley, 2023-04-11:
Hi thanks for opening this. I am busy with other things recently so apologies to the late reply. 

Since I don't have a k9.3 or a DM1.12, can you let me know what firmwares they are using by default? So I can better know what's the difference between the config. Thanks.

OKflashlightaholic, 2023-04-11:
I believe they both use the K9.3 tintramp hex, it only does turbo on the
fet channel. Does that help?

SiteRelEnby, 2023-04-15:
Newer DM1.12s use the D4SV2 firmware, at least for ones with the DD FET on the flood channel (not sure about SBT90 ones where the FET is for throw).

OKflashlightaholic, 2023-04-15:
Okay,  is it possible to use your d4sv2 hex but disable the 200% turbo?
Where when you double click for turbo it only uses the channel with the fet?

SiteRelEnby, 2023-04-15:
If you set your 2C style to "ceiling only" you will get level 130 in whichever channel is active, which is essentially disabling 200% turbo, as 130 is the default ceiling for the dual channel D4Sv2.

If you mean you want 2C to *always* use the FET channel for turbo regardless of previous channel, that'd need code changes, but wouldn't be that hard to do yourself.

You'd want to add a var to https://github.com/starryalley/Anduril2/blob/workspace_rev_647/spaghetti-monster/anduril/ramp-mode.c to store your tint mix, then when [going to turbo](https://github.com/starryalley/Anduril2/blob/workspace_rev_647/spaghetti-monster/anduril/ramp-mode.c#L213) you'd do something like:

```
saved_tint = tint;
tint = 1 ; //or tint=254 for the other channel - you may need to experiment to see which is which
set_level_and_therm_target(turbo_level);
```

and then replace the code [coming out of turbo](https://github.com/starryalley/Anduril2/blob/workspace_rev_647/spaghetti-monster/anduril/ramp-mode.c#L216) with
```
tint = saved_tint;
set_level_and_therm_target(memorized_level);
```

OKflashlightaholic, 2023-04-15:
Thank you, I unfortunately don't have the skills to do code changes to
anduril.

SiteRelEnby, 2023-04-15:
Here's a quick proof of concept with .hex files for 1 and 254 and source. Not tested but it's a 4 line change so "if it compiles. it works" in this case.

[starryalley_d4sv2-tintramp-fet_channel-specific-turbo.tar.gz](https://github.com/starryalley/Anduril2/files/11238112/starryalley_d4sv2-tintramp-fet_channel-specific-turbo.tar.gz)
[starryalley_d4sv2-tintramp-fet_channel-specific-turbo.zip](https://github.com/starryalley/Anduril2/files/11238115/starryalley_d4sv2-tintramp-fet_channel-specific-turbo.zip)

Probably not something that could be easily baked into it without adding too much code size, but as a DM1.12 fan I had to help :P

OKflashlightaholic, 2023-04-15:
Awesome, thank you so much! 😊

OKflashlightaholic, 2023-04-15:
I just tried both and both when double clicked both channels activate.

SiteRelEnby, 2023-04-16:
Ah, I didn't think about changing the default 2C style. Set your 2C style to "ceiling only" and ceiling to 150.

starryalley, 2023-04-17:
Thanks @SiteRelEnby for jumping in to assist. Yeah I agree that this needs some code change to do what you need but setting 2C style to ceiling only should work.