starryalley/Anduril2 #23: TS10 momentary 8C

Light-Veteran, 2023-04-08:
Hi! I am new here so i hope this is the right place, if not i am so sorry. This is the first time i use this app

I set floor at level 10 in smooth and doing 8C for momentary but the lvl i see is the default lvl i set memorised in ramp so not 10. 
In stepped mode 8C works correctly

starryalley, 2023-04-11:
Thanks for the report. However I thought 8C mementary mode uses whatever level is memorised as the level? This is from `momentary-mode.c`:

```
    // light up when the button is pressed; go dark otherwise
    // button is being held
    if ((event & (B_CLICK | B_PRESS)) == (B_CLICK | B_PRESS)) {
        momentary_active = 1;
        // 0 = ramping, 1 = strobes
        if (momentary_mode == 0) {
            set_level(memorized_level);
        }
        return MISCHIEF_MANAGED;
    }
```

So whatever your "memorized_level" is, it will use that. Not the floor level. But I never use momentary mode to be honest. Is this a different behaviour than the stock Anduril2?

Light-Veteran, 2023-04-11:
Thank you for reply. Ok I understand what happens. Its my fault and Anduril 2 works well and also in the same way with your firmware so I am stupid 😂
From off 1H is "moon lvl" and if you set 8C after 1H the momentary gives you the standard lvl you set. 
Viceversa if you switch on the light and ramp down to moon lvl and then set 8C the momentary give you the moon lvl. 
So momentary don't work as i aspected with 1H from OFF. Solved, no issues

closed 2023-04-11 by starryalley

Light-Veteran, 2023-06-19:
Hi Mark, sorry if I write here but I don't find in this app a right location for "private message" so I use this! 
Recently Durval Menezes 
add In custom firmware a function I asked for switch OFF and memorised the Aux RGB with 8C. I know 8C is for momentary and 8H is for middle tint. You think is possible to add this function? 
You think 3C is useful? Personally I don't add one level at time so if you want and you think is a good idea you can change 3C to smooth/ramp and original, 9C to switch off Aux and 3H to add 1 level. What you think? 
Cheers 
Simone