starryalley/Anduril2 #24: Wurkkos TS11 / TS25 support

tri4ng1e, 2023-04-14:
Hello!

Is it possible to add Wurkkos TS11 / TS25 support? They seem to share the same firmware, code `0715`. I found the config file [here](http://bazaar.launchpad.net/~gabe/flashlight-firmware/anduril2/view/head:/ToyKeeper/spaghetti-monster/anduril/cfg-wurkkos-ts25.h?sort=filename).

I do have TS11 (NLD today) shipped with version 2023-07-25-0715, but I don't have a probe yet. I ordered one from Hank, and I can help with testing when it arrives.

Thank you!

starryalley, 2023-04-17:
Yes I can include the build however I won't be able to test it as I don't plan to buy TS11/TS25 any time soon. But it looks like TS11/TS25 is T1616 so the flashing kit from Hank won't work on them?

starryalley, 2023-04-17:
Please give it a try. It's a direct copy and compile. If everything works fine I'll include in the next release.

[anduril.wurkkos-ts25.hex.zip](https://github.com/starryalley/Anduril2/files/11247051/anduril.wurkkos-ts25.hex.zip)

tri4ng1e, 2023-06-18:
@starryalley 
Finally, I received my parcel from Hank. As expected, I was able to flash TS11 with his probe and USB-UART module I already had. Happy to report that all functions seem to work well!

starryalley, 2023-06-18:
Thank you for the update. I've also included TS25 hex in the latest release here https://github.com/starryalley/Anduril2/releases/tag/release_20230614

closed 2023-06-18 by starryalley