starryalley/Anduril2 #25: Rgb switch off

Light-Veteran, 2023-06-21:
Hi Mark, sorry if I write here but I don't find in this app a right location for "private message" so I use this!
Recently, Durval Menezes
add in custom firmware a function i asked for switch OFF and memorised the Aux RGB with 8C. Every time we are in low with RGB we have to do 21C to switch off, is tedious. I know 8C is for momentary and 8H is for middle tint.
You think 3C is useful? Personally I don't add one level at time with 3C so if you want and you think is a good idea you can change 3C to smooth/ramp (like ToyKeeper firmware), 9C to switch off Aux and 3H to add 1 level. What you think?
Cheers
Simone

starryalley, 2023-06-21:
Hi Simone, yeah reddit is crazy in terms of messaging.

Thanks for the suggestion. I myself also find 21C very hard to get to, when I want to switch AUX from high to low. Also I am feeling I will eventually remove the `3C` in this fork because of the new multi-channel feature by Toykeeper. 

I'll see what I can do. Most likely I will keep whatever Toykeeper did in the new multi-channel branch, and then I will move around my changes to more clicks. `8C` or `9C` is a good use for controlling what you mentioned, or just to control the second set of AUX (like button LED alone).  There is also that tactical mode added in the main anduril2 branch that I still can't find time to include.

Currently I prefer to waiting for Toykeeper to finish from her side and then I will start moving my mod on top of that multi-channel branch, so this might have to wait a bit of time then...

Light-Veteran, 2023-06-21:
Thank you so much Mark. i'm in no hurry. I I'm here just to give the idea