starryalley/Anduril2 #27: TS25 doesn't have button light so shouldn't have all 6C modes

lukestein, 2023-07-31:
The cool "Allow the use of AUX/Indicator LED in lower levels (< DEFAULT_LEVEL) level (6C/6H while light is on)" setting cycles through four modes with 6C, but the middle two provide no illumination

starryalley, 2023-07-31:
Thank you for reporting the issue. I'll fix that.

starryalley, 2023-08-07:
Removed the 2 modes that are related to button LEDs. If you have time would you please test this hex for TS25. Since I don't have any anduril2 lights with RGB aux but without button LED.

Thanks.

[anduril.wurkkos-ts25.hex.zip](https://github.com/starryalley/Anduril2/files/12271459/anduril.wurkkos-ts25.hex.zip)

lukestein, 2023-08-07:
> [anduril.wurkkos-ts25.hex.zip](https://github.com/starryalley/Anduril2/files/12271459/anduril.wurkkos-ts25.hex.zip)

The fix seems to work perfectly on my TS25. Awesome!

I haven't tested everything else, but (as expected) at dim settings,
- 6C toggles mixed-in aux (high level) on/off
- 6H cycles through aux colors

starryalley, 2023-08-08:
Thanks for testing. Fixed by commit 5fb3634126df73128b6bc11465afac766bfb0173. Closing.

closed 2023-08-08 by starryalley