starryalley/Anduril2 #28: Feature request: Allow indicator light mix with candle mode (on no-aux TS10

lukestein, 2023-08-01:
My orange-indicator TS10 looks great in fireplace mode but the indicator doesn't work in default/stock candle mode. Also, the indicator is always at a low level when mixing into candle mode.

Feature request: On a TS10 in default/stock candle mode, have 7C toggle between indicator off, indicator low, and indicator high.

_Or_, have 7C toggle between indicator on and off, but use high indicator level if the main LED level is below some threshold.

Happy to discuss or ideate further!

Cf
> In candle wobble style (default/stock) we can additional use 7C to toggle if we want to use aux led to assist in tint mixing (if AUX is available)

lukestein, 2023-08-05:
Just realizing my suggestion may relate to [this commit](https://github.com/starryalley/Anduril2/commit/5bf19cbdd4e6a3d40020b3a7015f1645599dbb17)

(I of course agree that high indicator is too bright at low candle levels)

starryalley, 2023-08-07:
That's a good idea. I have 4 TS10's with blue/orange/red/green AUX, and high is sometimes too bright, and low is always too low. So I guess on a light with indicator LED only like TS10, 7C can be re-purposed to set indicator LED to off/low/high for all wobble style (default candle, fireplace slow, fireplace fast). On a light with AUX, nothing will be changed.

lukestein, 2023-08-07:
> So I guess on a light with indicator LED only like TS10, 7C can be re-purposed to set indicator LED to off/low/high for all wobble style (default candle, fireplace slow, fireplace fast). On a light with AUX, nothing will be changed.

This sounds to me like a great solution. I bet I'll wind up using white+bright orange, and white+dim red.

starryalley, 2023-08-08:
This is now implemented in commit 72fccd933209e6eba36bdedbe2a0b6670bdf3076.
Please give it a test if you have time. Thank you for this great suggestion.
[anduril.wurkkos-ts10.hex.zip](https://github.com/starryalley/Anduril2/files/12287447/anduril.wurkkos-ts10.hex.zip)

I have tested it on my 3000k TS10 with red aux.
Attached ts10 hex.

lukestein, 2023-08-09:
Thanks! I just tried the new release. Seems to work nicely with what may be a small bug:

Indicator mixes into candle if we reach candle directly from on → 3H

However, indicator mix doesn't work if we enter candle from other strobe modes (i.e., lightning → 3C or tactical strobe → 2C); in these cases we get standard white candle only

starryalley, 2023-08-09:
Thanks for testing and reporting. Yes it was my error, not resetting flag when doing `2C` or `3C` to switch strobe modes. I've updated the [latest release](https://github.com/starryalley/Anduril2/releases/tag/release_20230809). Appreciated your help here :)

lukestein, 2023-08-09:
Thank you! Your fix seems to have done the trick, and I haven't noticed any other issues with this updated release 🎉

Light-Veteran, 2023-08-09:
Don't worry, we try and test for you😉

starryalley, 2023-08-09:
Awesome thanks @Light-Veteran . sorry I haven't got to your requests. I'll get to them one by one!