starryalley/Anduril2 #30: Issues or probably a function in 3H?

Light-Veteran, 2023-08-18:
Hi Mark, thank you always for your work and custom firmware. I'm trying last release for D4SV2 tint ramp. 
When in Strobe modes (3H from Off) I notice the 3H (in some strobe modes) switch between the two channels but in smooth mode. For example in candle mode I can switch from cold channel to the warm with a mix or a middle tint and not with extreme tints only. 
Also there is not memory when you choose the channel. 
Cheers, Simone

starryalley, 2023-08-23:
Hi Simone, sorry for late reply. 

If I understand what you are saying correctly, yes in some strobe modes 3H is the same function as the normal ramp mode to change tint. I didn't change this behaviour. So if your current tint changing mode is smooth tint ramping (there is also stepped tint ramping), 3H will do smooth tint ramping in candle mode as well.

When you say "no memory when you choose the channel", do you mean if changing tint using 3H in candle mode, the chosen tint won't be remembered? Yes this is correct and deliberately done by me. Because in some modes in strobe modes the tint will be modified (tint alternating strobes, or fireworks mode for example). When entering strobe mode it will keep record of current tint and when exiting strobe mode it will restore that original tint, hence using 3H in candle mode to change tint, and then 1C to off, then 1C to on, the tint will still be the original tint.

You did make a good point. Maybe I should not restore the tint if it is changed by user changing it using 3H in any of the strobe modes, and will only restore it if it is changed by the strobe mode itself.

Light-Veteran, 2023-08-23:
Hi Mark, don't worry for delay. You reply when you can and is possible for you. 
Is clear and is correct what you choose is the best for the code you write.

This is the point
_**Maybe I should not restore the tint if it is changed by user** changing it using 3H in any of the strobe modes, and will only restore it if it is changed by the strobe mode itself._

For example.. I prefer warm tint in candle mode so when I use it I want to find the warm channel i select. So if the user change a parameter he can find immediately what he changed and he don't re do the same change every time. 
For example..
In candle mode (if start with cold channel) 3H add warm channel but if I click 3H another one nothing happens and not return to cold channel. So in this way I can't change what channel I prefer. 
I try 3H also in normal mode and it is strange and difficult to write here what the command do. 
I start always with only W2 (cold tint) and if I do 3H the flashlight goes in mix tint like 8H do, so is redundant?