starryalley/Anduril2 #31: Request: Support for new Wurkkos TS10 with RGB aux

lukestein, 2023-10-06:
The new Titanium TS10 has rgb aux and comes with a new build of Anduril (that includes e.g., a short battery check at shutoff).

At some point I'd love to get on a starryalley build ;)

starryalley, 2023-10-06:
Haha nice one! I actually received the TS10 Ti 2 days ago, and I can't run my own firmware there yet because I haven't migrated mine to be based on the new multi channel firmware by Toykeeper. 

I did play around with the built-in TS10 v2 firmware and then flashed her latest release too. It was awesome. 

I'll find some time to move some of my stuff on top of that. Can't promise it will happen soon because apparently the new firmware branch is very different and has many new stuffs. I hope I can find some time to do it soon.

lukestein, 2023-10-14:
Oh I guess I should try to flash the newest multichannel release to my v2, then. I wonder what's different from the version that came installed (which—how old could it be? The model just came out!)

starryalley, 2023-11-02:
Yeah definitely try the newest release from multichannel. It is great.

My TS10 Ti died over the weekend suddenly. It just won't respond to button presses. Using an good Alu TS10 head on the same Ti body worked so it was the head that died. Wurkkos is sending me a new light when the new batch arrives anyway, which is good.

Also reading that TK has just finished multi-channel branch and [will soon merge to the main branch](https://budgetlightforum.com/t/anduril-2/62656/1770), I don't think I will start making my own changes on top of the latest Anduril2 any time soon. I just don't have the time (and urge) yet probably because most of my lights are just single channels. 

Anyway I'll update when I start working on this.

starryalley, 2023-12-13:
Just an update: I've been slowly migrating my modifications on up of the latest Anduril code [here](https://github.com/starryalley/anduril). Currently only the following features are added:
- temperature AUX mode
- indicate low battery using AUX breathing mode
- configurable candle mode with 2 additional wobble styles (fireplace modes)
- add fireworks, lighthouse, broken fluorescent strobe modes

Since the AUX leds are now abstracted into channels for each color, it no longer works for all those indicator/AUX additions I made, for example allowing AUX on when main emitters are on in ramp mode and some strobe modes. I have to figure out a way to do that.

Anyway I am happy with my TS10 v2 with this current version. I don't think I'll add anything new to this repo then so I'll slowly move things up there.